AESD-Course-Project/AESD-Course-Project.github.io #3: Port Tensorflow 1.0 VisualWakeWords code to Tensorflow 2.0

ZachTurner07, 2021-03-29:
Develop a training-bed to port Tensorflow Code to detect persons using COCO - Dataset on local host.
Tensorflow setup is not needed on Arduino Board , so this setup has nothing to do with either the Jetson or Arduino Board.

DoD: Have a functioning repository to be added run wth a simple install script and --help flag.

Blockers: Not a lot of support for code porting and backlog checking. Original Code tested on Tensorflow 1.0. Which is backtracked. This would involve rigorous testing for success and tracking dependencies for setup
Linkers: Not needed to be ported on Jetson, so can be developed independent of Hardware.

arpit6232, 2021-04-01:
Added a repository as a submodule to the final submission : 
https://github.com/arpit6232/visualwakeup_aesd.git

closed 2021-04-01 by arpit6232

arpit6232, 2021-04-05:
Commits : 4d61a52 (visualwakewords)
Code under separate repository : https://github.com/arpit6232/visualwakeup_aesd.