AESD-Course-Project/AESD-Course-Project.github.io #4: Arduino Setup Script post build

ZachTurner07, 2021-03-29:
Create a shell script to pull Arduino-ide, arduino-CLI with necessary libraries to integrate TinyML Arduino Board and shield.

DOD: The Arduino Code should be tested to build on Host Machine and tested thoroughly to download and build the TinyML libraries and tensorflow bed to setup a script to install and set under $HOME/software

Blockers:

Setup script to be tested on Tegra/Jetson post Successful Yocto build for portability test.
Build Yocto Build and test on target.

arpit6232, 2021-04-01:
I feel this is slight duplication of the issue https://github.com/AESD-Course-Project/AESD-Course-Project.github.io/issues/6. 
Could the two be merged @ZachTurner07 .

arpit6232, 2021-04-04:
There seems to some libraries which are build not developed for ARM devices for the Arduino-CLI. The future plan would be to transfer code on the Arduino on X86 (host). Plug and play on Jetson.

ZachTurner07, 2021-04-05:
Does this issue describe updates to the Jetson or the host machine? I pulled this from your individual issues when I created it, and thought it was for a user to get the environment set up to interact with the Arduino. 

This seems more like an enhancement than a demo milestone, so that there's a one line setup command to be able to flash the arduino. However, if we have that done then we can for sure close it.

arpit6232, 2021-04-05:
https://github.com/cu-ecen-5013/final-project-arpit6232/issues/3. Opened a separate issue here.

CalebProvost, 2021-04-05:
End of Sprint 1.  
See continuation of issue for Sprint 2 as @arpit6232 stated above here: https://github.com/AESD-Course-Project/AESD-Course-Project.github.io/issues/11

closed 2021-04-05 by CalebProvost