AESD-Course-Project/AESD-Course-Project.github.io #6: Get Nano demo running

ZachTurner07, 2021-03-29:
DoD: Get demo of Arduino Nano running image detection running, and printing to the terminal

- [ ] Validate camera detected by Arduino (either shield or wired)

- [ ] Find source code to run on the Arduino to demo image processing

- [ ] Test Image detection, record accuracy

arpit6232, 2021-04-01:
Running Accuracy of 75 % currently. For more precision float values would be needed, arduino does not have an FPU. 
More optimization hangs the arduino.

Simultaneously testing to try to get a USB Camera (Webcam) started on the Jetson. 

Follow the instructions to setup the Arduino- IDE and Arduino-CLI to setup with the necessary libraries installed. 
Instructions to setup Hardware and Software here : https://github.com/AESD-Course-Project/AESD-Course-Project.github.io/blob/gh-pages/src/arduino_setup.md

closed 2021-04-01 by arpit6232

arpit6232, 2021-04-05:
Does not link to any specific commit. Is developed on a separate repository. And will be in the future added as a submodule.
https://github.com/arpit6232/arduino-library