AESD-Course-Project/AESD-Course-Project.github.io #16: Create Daemon and init script for uartserver

ZachTurner07, 2021-04-10:
now that a user application is working, it needs to be "daemon"ized if it makes sense. 

DoD: add a daemon feature to the UART server

CalebProvost, 2021-04-10:
Our yocto layer has been set up to install your daemon similarly to that of Assignment 6. I haven't tested it yet, but when your ready, the init script we made for A6 _should_ work to init your daemon as well.

ZachTurner07, 2021-04-17:
init script created, now we just need to run it auto on start up

CalebProvost, 2021-04-18:
As per the communication we had with Dan (see below). The init script process we were taught in this course isn't going to work...  
We're gonna need to update it to reflect a .service style daemon as referenced with the link provided.  

> 
> - Caleb Provost
> @Dan,
> Is the inherit update-rc.d method we learned in this course gonna work for OE4T, or is there a different method that is used compared to our current understanding & implementation?
> https://github.com/cu-ecen-5013/final-project-CalebProvost|cu-ecen-5013/final-project-CalebProvost
> 
> - Dan
> I expect it should work as we did in the class, but the best thing to do is always reference another recipe you are using with the same branch, so I often cross reference whatever I've done before with whatever other recipes in the same branches or used layers are doing
> Poky should have lots of examples
> Actually now that I think about it I think your default startup is going to be systemd
> so initscripts won't do anything
> and you need to look for a systemd startup example
> 
> - Caleb Provost
> Okay. I'll hunt down some systemd examples and see how to migrate our methods with that.
> Any good resources besides google/yocto manual?
> I'm not against it. Just figured I'd ask someone with the experience
> 
> - Dan
> The mender guys often have some good examples for this kind of stuff, I think I'd look at https://hub.mender.io/t/how-to-create-your-first-recipe-and-enable-auto-start-using-systemd/1195
> 
> - Caleb Provost
> Yup. I knew I should ask. That looks like a gold-mine. Thank you!

CalebProvost, 2021-04-19:
With @ZachTurner07's [uartserver.service](https://github.com/cu-ecen-5013/final-project-ZachTurner07/blob/develop/uartserver.service) code I updated our [yocto-recipe](https://github.com/cu-ecen-5013/final-project-CalebProvost/blob/ebf120e373d28ac123be6f0dcf2bf56cfae9f297/recipes-core/project-server/project-server_1.0.bb#L19-L39) to install the updated version as an initialized daemon.  

The latest image is current;y flashing, results pending...

CalebProvost, 2021-04-19:
Results of the latest changes weren't so good... The image is unusable with the init script adaption and causes it to no longer boot....

CalebProvost, 2021-04-19:
Init script adaption ended up blocking the image from functioning.  
Closing this issue as a blocker and reopened it as https://github.com/AESD-Course-Project/AESD-Course-Project.github.io/issues/29 in order to solve during next Sprint.

closed 2021-04-19 by CalebProvost