AESD-Course-Project/AESD-Course-Project.github.io #19: Start to Finish Build Integration

CalebProvost, 2021-04-12:
DoD: Merge all build tools and environments into one tool which can be used to execute our specific project as a whole.

- [ ] Uses docker build tool for OE4T: https://github.com/cu-ecen-5013/final-project-CalebProvost/tree/dl4t
- [ ] Uses docker build tool for Arduino: https://github.com/cu-ecen-5013/final-project-CalebProvost/tree/dard
- [ ] Adapts the tools to build SD Card image for Tegra while additionally flashing associated human recognition demo image onto the Arduino
- [ ] Docker pull, Docker run, flash SD Card, power-on and enjoy simplicity

arpit6232, 2021-04-13:
Not a biggie, but I got this error on a few trials on dl4t branch. 
[image]

CalebProvost, 2021-04-18:
> 
> 
> Not a biggie, but I got this error on a few trials on dl4t branch.
> [image]

Sorry for the delayed reply here, the Slack channel got this resolved. I'll update here for readers:  

This was due to docker group permissions needed to allow for file modifications from the container in a user permissions settings. Once 'docker' was added to the user's group and the permissions allowed for modifications, there was the added issue that the prior method of containerization of dependencies were contaminating the build directory (hence the `*.sock` files being in the build directory).  
The latest updates to the docker container resolves the prior issue where the post-installation steps being followed resolves the prior.  

I will continue working on integration of these fixed and updated tools today.

CalebProvost, 2021-04-19:
With the excessive debugging with the blockers I had on other issues/tasks, this tasks has been pushed to the next sprint.  

Current progress timestamp: https://github.com/cu-ecen-5013/final-project-CalebProvost/commit/a5f5b431ea97731559cf300927f99c0a06cdaba7 

Continuation sprint task/issue: https://github.com/AESD-Course-Project/AESD-Course-Project.github.io/issues/28

closed 2021-04-19 by CalebProvost