AESD-Course-Project/AESD-Course-Project.github.io #24: API , Setup and Response.

arpit6232, 2021-04-15:
DoD : Create an API between Jetson Nano and response from the Arduino BLE 33 Sense. 

Blockers: Testing, Arduino State machine response time to ARDUINO CLI.

arpit6232, 2021-04-15:
As per @ZachTurner07 , 
UML Looks like this : 
[image]

arpit6232, 2021-04-16:
Tested over Host, 

Commit : [a84a54d1917d7df28d2f5b73ebd9d51594dba393](https://github.com/arpit6232/arduino-library)

- Prerequisites
```
sudo stty -F /dev/ttyACM0 cs8 9600 ignbrk -brkint -icrnl -imaxbel -opost -onlcr -isig -icanon -iexten -echo -echoe -echok -echoctl -echoke noflsh -ixon -crtscts

sudo chmod 777 /dev/ttyACM0
``` 

- [image]

- [image]

arpit6232, 2021-04-17:
Tested the above Functionality on Jetson Nano, Arduino BLE reacts correctly.

ZachTurner07, 2021-04-17:
Going to integrate this testing with UART server to validate integration

ZachTurner07, 2021-04-18:
Trying to get the API to work with the user application

ZachTurner07, 2021-04-19:
Simplifying API from words to single bytes simplified the implementation. Also, bytes are echoed back instead of responding with a different message to confirm it was detected.

ZachTurner07, 2021-04-19:
commit [b78b2e5604f8d9c093e8094058d9f11884ed6531] establishes 2-way communication (https://github.com/cu-ecen-5013/final-project-ZachTurner07/commit/b78b2e5604f8d9c093e8094058d9f11884ed6531)

ZachTurner07, 2021-04-19:
Jetson Nano can send a serial command to the Arduino to command it to start

closed 2021-04-19 by ZachTurner07