AESD-Course-Project/AESD-Course-Project.github.io #26: Integrate yocto package for arduino_cli to meta-final-project

arpit6232, 2021-04-18:
DoD: Integrate PR for [arduino_cli](https://github.com/cu-ecen-5013/final-project-CalebProvost/tree/arduino_cli_package) into [yocto-layer](https://github.com/cu-ecen-5013/final-project-CalebProvost/tree/yocto-layer)

Blocker: 
[image]

arpit6232, 2021-04-18:
Adding @CalebProvost in since yocto source code is managed on the Dockerfile

CalebProvost, 2021-04-19:
> 
> 
> Adding @CalebProvost in assignes since yocto source code is managed on the Dockerfile

It's not managed in the Dockerfile... The repo with the source code is found here:  
https://github.com/cu-ecen-5013/final-project-CalebProvost/tree/yocto-layer
The Docker container made by the Dockerfile just contains all the dependencies to make the yocto image. The `build.sh` script is what pulls and executes the build.

arpit6232, 2021-04-19:
Pull request under review : 
https://github.com/cu-ecen-5013/final-project-CalebProvost/pull/1

CalebProvost, 2021-04-20:
@arpit6232 has tested his recipe with our Distro and confirms it works. I've resolved the merge conflicts and send further testing back to him.  
In addition to final confirmation that the build continues without issues after merge, @arpit6232 is additionally testing [this commit fix](https://github.com/cu-ecen-5013/final-project-CalebProvost/blob/b7b8904cc96e3ec224486e028dfc703ac8607f60/recipes-core/project-server/project-server_1.0.bb#L25-L39) for https://github.com/AESD-Course-Project/AESD-Course-Project.github.io/issues/29

arpit6232, 2021-04-20:
Arduino-cli integrated and paths defined to work with uart-server.

closed 2021-04-20 by arpit6232