AESD-Course-Project/AESD-Course-Project.github.io #29: Resolve SystemD Adaption Issue

CalebProvost, 2021-04-19:
Continuation and Resolution Task for the blocker of issue https://github.com/AESD-Course-Project/AESD-Course-Project.github.io/issues/16
---
Original DoD: add a daemon feature to the UART server  

Blocker: Yocto image yields no response from board  

New DoD: 
- [ ] Evaluate boot log (attach to boot serial and capture printk logging)
- [ ] Adjust either the [source code](https://github.com/cu-ecen-5013/final-project-ZachTurner07/blob/develop/uartserver.service) or the [yocto recipe](https://github.com/cu-ecen-5013/final-project-CalebProvost/blob/ebf120e373d28ac123be6f0dcf2bf56cfae9f297/recipes-core/project-server/project-server_1.0.bb#L19-L39) to resolve the boot issue

CalebProvost, 2021-04-20:
Latest changes to the recipe:  
https://github.com/cu-ecen-5013/final-project-CalebProvost/blob/b7b8904cc96e3ec224486e028dfc703ac8607f60/recipes-core/project-server/project-server_1.0.bb#L25-L39  

Handed it off to @arpit6232 for further testing as his implementation of a systemd.service is working.... and it takes nearly 3 hours to flash an SD card for me...

arpit6232, 2021-04-20:
Upon SD Card flash, uartserver.service detected and code uploaded to begin work on the Jetson Nano. 
[image]

Debugging through to check if after poweroff, the serial service and restart

CalebProvost, 2021-04-21:
It turns out this is probably related to https://github.com/OE4T/meta-tegra/pull/665/files  
I've added that definition in the distro config file and am testing now to see if it'll update with the correct layout.

CalebProvost, 2021-04-21:
Turns out, there was an update with the `meta-tegra` master branch which switched the method the SD Card images are created. As such, the method I was using and expecting has all the boot partitioning and information within the SD Card image itself.  
This is no longer the case. @arpit6232 has already performed the FRC on-board flashing method, which was where our confusion was derived from.  

We'll update documentation to demonstrate that the boot information needs to be updated on the board 1st, then these images (ours and the newest one's from meta-tegra) will work after flashing the SD Cards.

closed 2021-04-21 by CalebProvost