AESD-Course-Project/AESD-Course-Project.github.io #30: Integrate i2c-tools to yocto image

arpit6232, 2021-04-20:
DoD: Successfully detect i2c-tools on the yocto image to function in parallel if possible to GPIO's
Blockers: tools present under yocto, but iamge does not include it.

arpit6232, 2021-04-20:
Source Code : https://github.com/cu-ecen-5013/final-project-CalebProvost/tree/yocto_layer_fix_With_i2c
Hardware Integration (Ambitious Goal and hardware dependent): https://github.com/arpit6232/JHLEDBackpack, successfully tested the above functionality on host. 
This is an added functionality and might not be implemented in the final image.

arpit6232, 2021-04-21:
This project would be considered as part of future goal and would not be integrated with uartserver.

- I2C based seven segment displays have taken longer to get delivered, and would not be available with everyone in the team before the demo-day. 

Code worked upon : https://github.com/cu-ecen-5013/final-project-arpit6232/tree/main/jetson_display
Code Test details: https://github.com/AESD-Course-Project/AESD-Course-Project.github.io/issues/23.

CalebProvost, 2021-04-22:
I'm building and flashing this image now to provide secondary confirmation that the i2ctools are provided in the recipe @arpit6232 has created.  
I expect to provide feedback by EOD tomorrow.

arpit6232, 2021-04-23:
Tested the i2ctools on Jetson, included from the image. Closing the issue.

closed 2021-04-23 by arpit6232